makerbase-mks/MKS-SGen #2: Smoothie firmware

pmth, 2019-06-16:
Hello,
If I would like to use smoothieboard firmware, should I use firmware and configuration from SBase (since I only see Marlin firmware here)? How do I configure it to use TMC2206? Thanks!

pmth, 2019-06-17:
I figured this my self. TMC2206 seems to be configured by the jumper under the unit and adjusting Vref for current limit. UART is probably not supported. I use smoothie firmware, downloaded from smoothieware site, and configuration from sbase repo with some modification to what I need as I am using it for K40 laster cutter.

closed 2019-06-17 by pmth

OlegIvanus, 2021-07-18:
Maybe this would be actual for new users: 
1. You need to install VS Code
2. Install plugin for VS Code named PlatformIO, 
3. Download Smoothieware sources from GitHub and open this folder from PlatformIO
4. Find and open configuration.h file
5. Find in configuration.h strings "#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209"