makerbase-mks/MKS-SGen #4: Botloader hosed

xeddog, 2019-10-20:
I have the MKS SGEN L 32-bit controller for my 3d printer.  In the process of getting it properly configured, it looks as though the boot loader is hosed.  When I put a new firmware.bin file on the SD card and power cycle the board, it does not get converted into firmware.cur.

makerbase-mks, 2019-10-22:
You can format the TF card, then copy the firmware.bin to the TF, insert it into the SGEN_L motherboard, reset or power it back on.

xeddog, 2019-10-22:
I have done that several times with a couple of different cards.  With a freshly formatted card, firmware.bin copied onto it, inserted into the card reader on the board, and power the board, there is no firmware.cur created.  So either the bootloader is hosed, or the card reader itself is hosed.

kusuprfn, 2019-11-06:
I have had the same issue. 

It appears this marlin build hoses the bootloader and/or disables the SD card on the board. You can get it to flash if you revert to the smoothie firmware bin in this repository, (using the reset button to restart the board) then remove any connected screens with an SD card reader and use the reset button after inserting the SD card, but it still does not resolve the overall SD card issue once flashed. 

Once Marlin loads to the board LEDs 2,3 and 4 never turn on and LED 1 flashes. I get directional control on the steppers with it loaded this way, but have not tried anything else.

Granted, I've only been at trying to get Marlin to work for a day now, but this is the only way I've been able to get marlin to load to the board (kind of).

xeddog, 2019-11-06:
I forgot I had this posted here so I will update.  I got my board working.  I'm not sure WHY it started working, but it is.  During the time it was not working, I also noticed that the status LEDs on the board just weren't working the way they had been.  It's tough to explain here, but they were not lighting properly.  I had wasted an entire day trying to get it working, but finally gave up and started working on using a REARM/RAMPS 1.4 controller. 
About a week later I had this MKS board in another room at the desk where I do all of my Marlin compiling.  I saw the SD card that had the MKS firmware on it, so on a whim I inserted the card into the card reader on the board, and connected the board to the computer via a USB cable.  I watched as the LEDs on the board did their thing.  Then ... Wait!  What?  That looked like it worked!!!  When I unplugged the board and loaded the SD card into the computer, it had the firmware.cur on it.  I re-installed the board into the printer and it has been working perfectly since, firmware updates and all.  

Maybe it had something to do with powering the board with 5V USB instead of 12V psu?  Or more likely maybe the board just decided to stop screwing with me.  Whatever the reason, I have learned to stop questioning this printer.  It is a LOT meaner than me.  :-)

closed 2019-11-06 by xeddog