makerbase-mks/MKS-SGen #6: MKS SGEN 1.0 SD does not show on MKS TFT, Marlin 2.0

rpweaverjr, 2020-05-23:
Converted my Ender 3 Pro to 32 bit MKS SGEN v1.0 (not SGEN-L). Upgraded the display to MKS TFT 32. The drive on the TFT is available but the drive on the motherboard does not display on the TFT.  SD Support is enabled in Marlin 2.0. What else needs to be set in either firmware? Seems it works on Big Tree Tech products but i haven't found anything for MKS. I have the printer working and have successfully printed several items. 
I would also like to know how to connect/activate the UART interface on my TMC 2209s with this board, wiring and configuration settings. 
Thanks.

makerbase-mks, 2020-06-06:
MKS TFT connect to SGEN board by serial, MKS TFT read SD only of TFT board, does not read to SD on the SGEN board.

rpweaverjr, 2020-06-06:
Why not,  there is G code that reads the sd card on the sgen board? The TFT should send that G code and display the results.

doug-scott, 2020-07-08:
> Converted my Ender 3 Pro to 32 bit MKS SGEN v1.0 (not SGEN-L). Upgraded the display to MKS TFT 32. The drive on the TFT is available but the drive on the motherboard does not display on the TFT. SD [...]

Did you ever find out how to get your TMC2209s configured on your MKS SGEN LPC1769 board?  I tried both BigTreeTech and Makerbase but all I got were two people pointing at each other.

rpweaverjr, 2020-07-08:
They are currently working but not in UART mode. I just configured them in Marlin. I don't have the knowledge to know how to hook UART up. I could probably figure out a way to do it but it would take a lot of learning and understanding and documentation on how these circuits work to do it.

doug-scott, 2020-07-10:
By "configured them in Marlin" are you talking about the #define X_DRIVER_TYPE  TMC2209 entries for x, y, z and e?

rpweaverjr, 2020-07-11:
That's correct. [Image]

rpweaverjr, 2020-07-11:
BTW, they are very quiet.  I hardly know the printer is working (or finished)  unless I look at it. I have no complaints.

rpweaverjr, 2020-07-11:
I had to wire the extruder cooling fan and case fan directly to the power supply port on the board. There is no 24V connector on the board that i could find. That means these fans are always on when the power is on.

doug-scott, 2020-07-18:
My x-axis stepper is silent, my y-axis hums.  Its mounted in one of those plastic adaptors so it can be mounted to the frame.  I think that is creating the noise.  I have one of those rubber isolators on both a and y steppers, x is mounted directly to the carriage so the isolator actually works.  My board has a fan connection that has no problem driving the fans.  It only has the one port though and does not take a connector.  Bare wire only. The sGEN_L boards have a spare port with un-switched power on it.  No such port on my SGEN board.  I find that odd.

rpweaverjr, 2020-07-18:
The connector marked "fan" i used for the part cooling fan since the G code regulates that fan. It works add expected.