makerbase-mks/MKS-SGen #8: Unable to view SD card in motherboard from computer

dscott22099, 2020-06-08:
I am trying to build and upload the firmware from Visual Studio Code with Platformio.  I could see the SD card when connected via USB in Windows Explorer, and it showed the volume name (REARM), as well as the contents of the SD card.  Now I can see a drive letter, but it asks me to insert a SD card when I try to open it.  Can't put a card in, there is already one in there.  This used to work just fine a month ago, and it stopped working a couple of weeks ago.  I have followed all the instructions as to the max size of the primary partition and formatted as FAT32 and the volume name set to REARM.  Visual Studio Code can only see the SD card if it is inserted in my computer.  You wouldn't think it would be too much trouble to just put the card into my computer and build the firmware, but, I have the motherboard inside a case that requires me to completely disassemble the case just to be able to get at the SD card.  I have tried multiple computers, even did a clean install of windows to make sure it was not my particular windows environment.  Why can't I see the SD card in Windows anymore?

makerbase-mks, 2020-06-09:
Hi! Can you retry installing the smoothie USB driver?

doug-scott, 2020-06-09:
I downloaded and installed "smoothieware-windows-signeddriver-v1.1.zip" from their website.  I am running Windows 10 and I installed it via Device Manager.  I now have a com port called "Smoothieboard DFU Interface (COM 4)" when I plug the USB cable in, and it behaves the same way.  On their website they say Windows 10 does not need their driver.  Is that the driver you are asking about?

doug-scott, 2020-06-12:
I have also tried this on a new motherboard (MKS SGEN LPC1769) and am also unable to see the SD card when inserted in the motherboard.  The other board I am using is a MKS SGEN_L LPC1768.

makerbase-mks, 2020-06-13:
@doug-scott You use Marlin 2.0 firmware?

doug-scott, 2020-06-14:
Yes I do.  I have figured part of it out though.  This was working back in April, and stopped working in May sometime.  I found an archive of Marlin 2.0 BugFix release from the first week in April, so I unzipped it, put my current configuration.h and configuration_adv.h into the Marlin folder.  Made the changes to the one configuration and the platformio.ini for LPC1769 and compiled the firmware.  Flashed the MKS SGEN with it, and now I can see the SD card again from my PC.  Did the same thing for my MKS SGEN_L board, and I can now see the SD card from that printer.  

To me it acts like there was a change made that either broke it(bug) or it was an intentional change (new feature).  For users like me this change was not thought out completely.  My boards sit inside a case that requires disassembly for anything that requires access to the on board SD card.  Marlin 2.0 requires an SD card to flash the firmware.  By not being able to access it via USB anymore, it have to tear the box down just to be able to get that tiny card out, with old retired auto mechanic fingers.  Found out today that using the tip of your finger nail to push the little card in will fire the card a good 10-12 feet out of the room into the hallway if your nail slips.  I maybe 61 but I can still beat the cat at getting the card before she did.

doug-scott, 2020-07-07:
I downloaded and built bugfix-2.0.x and I can now see the SD card.  Don't know when it was fixed, I was using source from April because I knew it was working.  Also found we can set firmware to be case insensitive, again not sure when it was added, but it is there in configuration_adv.h.  They may want to set it to be active since Marlin is supposed to be case insensitive.

makerbase-mks, 2020-07-08:
So, this phenomenon is caused by firmware, caused by version.

closed 2020-07-08 by makerbase-mks