makerbase-mks/MKS-SGen #10: P1.23 as Servo

lps83, 2020-08-24:
I'm trying to use my BLtouch, but it's impossible.
I connected my oscilloscope on the pin P1.23, when I try M280 P0 Sxx ( different values ) nothing happen. 

Any ideas ? 

`//
// Servos
//
#define SERVO0_PIN                         P1_23  // J8-3 (low jitter)
#define SERVO1_PIN                         P2_12  // J8-4
#define SERVO2_PIN                         P2_11  // J8-5
#define SERVO3_PIN                         P4_28  // J8-6`

makerbase-mks, 2020-08-25:
Hi! Can you provide your configuration.h file and pin file?

doug-scott, 2020-09-08:
I am having the same issue.  The 3dtouch just flashes amber/red/orange colour all the time.  Online it says that means it has an error.  I used the "doesn't do anything" logic to say it has an issue.  I am running the MKS SGEN LPC1769 mainboard, with the 3DTouch sensor.  I have followed the online guides as best as I could, but, there is precocious little with this mainboard and any of the MKS accessories.  I have attached my config files.
[Configuration_mks-sgen1769-3dtouch.zip](https://github.com/makerbase-mks/MKS-SGen/files/5190572/Configuration_mks-sgen1769-3dtouch.zip)

makerbase-mks, 2020-09-09:
Use 3D TOUCH for MKS SGEN is same as MKS SGEN_L motherboard, I has check your configurati.h file and i find it have not config 3D TOUCH partment and so on, i think, you can refer to : https://www.youtube.com/watch?v=r2A0b7XQEaI
https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/3D_TOUCH

doug-scott, 2020-09-09:
> Use 3D TOUCH for MKS SGEN is same as MKS SGEN_L motherboard, I has check your configurati.h file and i find it have not config 3D TOUCH partment and so on, i think, you can refer to : https://www.youtube.com/watch?v=r2A0b7XQEaI
> https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/3D_TOUCH

The MKS SGEN board does not look like any of the _L boards.  Not even close to the same shape.  I have attached picture of both the actual board, as well as the pin assignment picture.  I am not understanding your comment of my configuration not having the changes for bltouch enabled.  They are there, were you looking for 3dtouch or bltouch?  The configuration files do not use 3dtouch, they use bltouch.  As far as I can tell I have made all the changes that have been suggested in all the different videos for bltouch installing and enabling.  

[image]
[image]

lps83, 2020-09-09:
Works for me now. Just don't use the UART for tmc220x and that will solve your problem. I'm going to switch to the Sgen L.

doug-scott, 2020-09-09:
> Works for me now. Just don't use the UART for tmc220x and that will solve your problem. I'm going to switch to the Sgen L.

I have the MKS TMC2209 and they just insert without anything more than setting the DRIVER_TYPE to TMC2209.  Can you post your config files including the pins file please?  And where did you connect the 3 wire connector?

lps83, 2020-09-13:
> I have the MKS TMC2209 and they just insert without anything more than setting the DRIVER_TYPE to TMC2209.  Can you post your config files including the pins file please?  And where did you connect [...]

Yeah that is the problem. Config as regular driver. That will solve your problem.

doug-scott, 2020-09-13:
I think I have finally figured out the 3DTOUCH hardware connections for my MKS SGEN LPC1769 v1.0 mainboard.  Note that this board is NOT a _L type board.
The 3DTOUCH comes with a wiring cable that connects to the PROBE then splits off into 2 connectors, one 2 pin and one 3 pin.  

1.  Take the RED middle wire out of the 3 pin connector, then connect that connector to the first 3 pins of the J8 header(reading left to right) as shown in the attached image.  Note that the GREEN wire connects to the first pin, and the YELLOW wire connects to the P1.23, there is no wire in the middle position.

2.  Take the 2 wire connector and plug it into the Z- header(the Z axis endstop connector on mainboard) with the YELLOW wire connected to the P1.28 pin and the BLACK wire to the centre pin marked as GND.  
3.  The RED wire you took out of the 3 wire connector is then connected to the third pin marked as 5V.  

You can just connect the bare terminal to that pin.  On mine I used to use an inductive sensor/proximity sensor that had a 3 wire connector that plugged into the Z- connector.  I cut it off and spliced the the 3 wires to it.  Just by chance the wires in that connector happened to be BLACK/WHITE/RED that matched the colour and position of the connector.

That got my 3DTOUCH sensor to cycle twice on power up, and fixed the flashing amber I was getting before.  I applied the changes to CONFIGURATION.H and the CONFIGURATION_ADV.H files.  I have not run the printer yet, I am waiting for a couple of parts that are preventing me from using the printer beyond connecting the TFT to my PC via USB.
[image]

makerbase-mks, 2020-09-14:
So, your bltouch is works when wiring is OK?

lps83, 2020-09-16:
> So, your bltouch is works when wiring is OK?

For me, it's the BLTouch or the UART for the TMC2209 ...

doug-scott, 2020-09-16:
> So, your bltouch is works when wiring is OK?

As it stands right now, the 3dtouch cycles twice then stops with the blue and amber LED on solid when powered up,  I haven't got the printer completely back together enough yet to see if it will work properly.  Right now the "all in one" box that contains all the electronics looks like it exploded with all the wiring connected but not sorted. I am expecting the LCD12864 tomorrow.  Once I see how it will connect, I will sort/route the wiring and see if it will work.

makerbase-mks, 2020-09-16:
@doug-scott 
You can try to use command control it, for example, M280 P0 S10 and so on

doug-scott, 2020-09-16:
> @makerbase-mks
> You can try to use command control it, for example, M280 P0 S10 and so on

When I enter that command nothing happens.  Use pronterface returns nothing in the log, and using S3D returns OK to the command, but does nothing with it.  I can manually move the printhead all around the table.  I have attached my current config files, I am not sure what you meant by "i find it have not config 3D TOUCH partment" last week, and I went through that document you provided the link for as well.  Currently I have no issues building the firmware, and the machine starts and responds to commands other than the M280 P0 S10 as suggested.  When powered on it cycles the probe twice, then sits there with both LEDs on solid.

The video link applies to two families of motherboards, neither of which are compatible to mine.

[Configuration0916.zip](https://github.com/makerbase-mks/MKS-SGen/files/5233660/Configuration0916.zip)

doug-scott, 2020-09-16:
forgot to include the other config files I editted.  This one contains all of them.

[config0916.zip](https://github.com/makerbase-mks/MKS-SGen/files/5234699/config0916.zip)

makerbase-mks, 2020-09-17:
@doug-scott 
Can you disenable #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN  and test it?

doug-scott, 2020-09-17:
Disabled that line, rebuilt firmware, flashed new firmware, sent M280 P0 S10 and I get the same result, probe does not deploy.  Still have both LEDs on solid.
I noticed that the pins_MKS_SGEN.h has a section that says 

#define MKS_HAS_LPC1769
#include "../lpc1768/pins_MKS_SBASE.h"

So I made the changes that should have been in the pins_MKS_SGEN.h in the pins_MKS_SBASE.h.  For whatever reason the developers only put the TMC_UART pin assignments in the MKS_SGEN file.  The config files already posted contains those two files.  Perhaps there is something else that needs to be set in the pins_MKS_SBASE.h file.  The MKS_SBASE motherboard looks identical in layout to the MKS_SGEN board from a strictly visual view.

lps83, 2020-09-17:
> forgot to include the other config files I editted.  This one contains all of them.
> 
> 
> 
> [config0916.zip](https://github.com/makerbase-mks/MKS-SGen/files/5234699/config0916.zip)
> 
> 

Do you want my config ?

doug-scott, 2020-09-17:
> 
> Do you want my config ?

What motherboard were you using before switching to the SGEN_L?  Can't hurt to see your config files.  Could you please include all the config files you altered?

lps83, 2020-09-17:
> > 
> 
> > Do you want my config ?
> 
> 
> 
> What motherboard were you using before switching to the SGEN_L?  Can't hurt to see your config files.  Could you please include all the config files you altered?

Im still using my SGEN. But I ordered a SGEN_L because I want to be able to use the uart for the TMC2209 AND the Bltouch.

doug-scott, 2020-09-17:
Yes, please send me your config files.  Could you also tell me where you made your connections on the motherboard?  It would be greatly appreciated.

lps83, 2020-09-19:
[image]

[Configuration.h.zip](https://github.com/makerbase-mks/MKS-SGen/files/5250517/Configuration.h.zip)

As you can see :
Servo side : Signal on P1.23 ground on the same connector than the P1.23 ( 2P left ), 5V on the Z endstop connector
Switch side : on the 2 remaining pin for the endstop connector.

doug-scott, 2020-09-19:
After more playing around I have found some things.  With the 12864 installed, I am seeing an error on the info screen that says TMC CONNECTION ERROR.  When I run a M122 from a pc it returns:  
SENT: m122
READ: Testing X connection... Error: All LOW
READ: Testing Y connection... Error: All LOW
READ: Testing Z connection... Error: All LOW
READ: Testing E connection... Error: All LOW

The printer has 5 MKS_TMC2209 installed, with no jumpers changed, or jumper wires installed.  I bought the MKS drivers to be sure there would not be any compatibility issues.  Actually every add on accessory has come from MKS, including the MKS_FES, MKS_OSC, MKS_TFT35, MKS_LCD12864, and MKS_DET.  It appears that thinking all components will remove working together issues does not apply to MKS.  

I have removed the MKS TMC2209 drivers from the motherboard, and removed the driver type definition from configuration.h file, then rebuilt the firmware. 

After restarting the mainboard to complete the flash procedure, connect with PrintRun, ran M502 followed by M500, and then
ran M122, and got invalid command, which I kind of expected since there are no drivers to check.  I then ran M119.

SENDING:M119
Reporting endstop status
y_min: open
z_min: open
z_probe: open

Where did x-min go?  If I try to home the printer, the z-axis does the little drop to ensure the head is not touching the bed, then homes the y- axis then it sends the printhead to the x-axis home where it just ignores the endstop switch and tries to keep going.  I can see that it simply does not know of the x_min endstop switch.  I checked the continuity of the switch and wiring from the x-min connection on mainboard, and the switch shows continuity when released, and no continuity when activated, so the switch and wiring are working correctly.  

The 3dTouch probe can now be activated with M280 commands, but still sits with both LEDS on when in the Stowed position, with only the blue LED on when extended.  

I recently purchased this board, and now I see that MakerBase has released a v2.0 of the MKS_SGEN_L board that has the LPC1769 chipset, and also many other desirable features like PWN fan support, dual z-axis connections(although that now means and extra driver will be required), multiple 12/24v ports, etc.  If that board will support all the accessories without having to spend days/weeks trying different configurations, it may be worth the money.  Then again, if it is something that Makerbase has not even tested with their own accessories(in other words, leaving all QA/QC/Documentation testing up to customer to perform) it will only be worth considering if MakerBase gives free upgrade boards to customers that are stuck weeks into trying to get the MKS SGEN board to work with what is supposed to be compatible hardware.

doug-scott, 2020-09-20:
> [image]
>
> [Configuration.h.zip](https://github.com/makerbase-mks/MKS-SGen/files/5250517/Configuration.h.zip)
>
> As you can see :
> Servo side : Signal on P1.23 ground on the same connector than the P1.23 [...]

My 3dprobe is connected the same way.  I went through your configuration.h file and it was essentially the same, few differences that made no change to my setup.  Could you post the other config files that you made changes in please?  The only way I could get rid of the TMC CONNECTION ERROR was to remove the drivers and disable the entries in configuration.h file.

Did you order the [this ](https://www.aliexpress.com/item/4001283346331.html?spm=a2g0o.cart.0.0.66e93c005ym2hy&mp=1)MakerBase MKS SGEN_L V2.0 board?  I would hate to pay for another board only to find they didn't actually verify that their hardware works on their boards properly.  The customer should not be their testing department.

lps83, 2020-09-22:
I will check my endstop status. 
Sadly I ordered the V1 of the Sgen L, but it was on amazon, so I'm just going to send it back.

I really disappointed by MKS. I have so many issues with the TFT35 too. nothing works properly. 
About the config, I didn't change any setting in the config_adv.

doug-scott, 2020-10-03:
New board arrived from Aliexpress yesterday, 11 days from order date to Toronto Canada, tough to beat time like that.  Now we can start over.

First thing I did was download bugfix source due to fixes to this being newly released hardware, there will likely be issues being corrected pretty often.  I unzipped to a clean SSD that only has this source code archive and completely clean config files.

I made the following changes to see if it can at least build the firmware.

Set  #define MOTHERBOARD BOARD_MKS_SGEN_L in configuration.h
Set  default_envs = LPC1769 in platformio.h

That does not build.  Gave an error about was I sure it wasn't 1768 or something like that. Took a look in the pins folder to see if they had changed the way they had it set up for the SGEN board by having a 1769 pins file that included the 1768 file.  They changed it.  Now there is a pins_MKS_SGEN_L_V2.h file.  Took a look inside it and saw this line:

#define BOARD_INFO_NAME   "MKS SGEN_L V2"

Since the build failed with what I thought was the correct motherboard name I figured I would just change the name to match that name, but not being a programmer, I did not include all the underscores to remove spaces and it failed miserably.  Changed the line to 

#define MOTHERBOARD BOARD_MKS_SGEN_L_V2

and ran the build again.  Now it builds without error, and uses the 1769 version of src files.  

I took my printer down to the frame and am going to only put it together once this time.  Switching over to GX16 connectors to clean up the wiring inside the box, allowing me to assemble the box completely on the desk and then just plug the harnesses in as needed, and be able to simply pull a connector to disable something.  Using 4 and 8 pin connectors.  There is something like 60 wires going into the box, with majority going to the mainboard.  24 wires alone for stepper motors.  It will take a bit to solder all the connections, and correct all the wire lengths.  Once that is done, I will do the 3DTOUCH first since it is needed to home the printer.  That should allow me to at least put some filament on the bed.  After that I will be enabling/adding in the other parts like TMC2209, TFT35, 12864, filament detector etc. one piece at a time.

Should I start a new thread for this?  I have a feeling lps83 will be along shortly with his new motherboard, assuming he ordered the same one I did.

lps83, 2020-10-13:
Good you got a V2. I sent back mine, waiting for Amazon to sell the V2.
I have the TFT35 and so many bugs too.
i should've stayed with my Atmega ^^

doug-scott, 2020-10-14:
I have given up on trying to use the GX16 connectors, the plastic they use to retain the pins melts at too low a temperature and the 8 pin connectors soften up too much by the 3rd pin to remain useful.  I have moved to standard JST connectors.  It is a birds nest of wires outside the box, but, all connections are labeled on either side of connector so it is easy to unplug it all to move the box to my desk when needed.  I am getting this back together now, but seems I am still getting similar results.

The 3Dtouch is behaving the same way with this board.  The board now matches the instructions in the video, except for the source file names that need to be edited.  I have set the pins file to use pin P1_25 and have connected the wires to the Z- port on motherboard.  If I try to home z-axis, the probe does not deploy, and the bed does not stop.  I have not got the offset set because I have no way to show the system where 0 is in the first place.

I just do not understand why this is not working.  Am I misunderstanding how the 3Dtouch is supposed to work?  In my mind the 3Dtouch should deploy every time the system is sent G28 or G29.  On my system it does not deploy.  I can manually deploy it via the 12864 interface.  When the system is powered up, the probe cycles twice.  I am certain that I have read that it is supposed to cycle 3 times, but, cannot find where I read that.  

I am getting the impression that manufacturers of these parts feel it is so easy to install and configure that they no longer need any proper documentation.  I have yet to find any videos or online documentation that actually applies 100% to the parts I have.  I understand that this is a continuous evolving system, but, if during QA testing they just update the documents, those 10 minutes would save them hundreds of man-hours in support work.  I doubt that I am the only one that thinks that no instructions means no configuration is required.  Just plug in the part and it should work.

doug-scott, 2020-10-17:
I finally got most of the issues sorted out with this new board (MKS SGEN_L v2.0), but, I get an error showing up on my 12864 that says "TMC CONNECTION ERROR.".  According to the support person on their Facebook site I have to remove or disable the #define X_DRIVER_TYPE lines for all 5 entries in configuration.h.  The error does go away, but, I have no idea what that impacts.  Waiting for a reply from support.

I finally got the 3DTOUCH to work.  Seems I was trying to do something that people with 3DTOUCH don't expect the sensor to do.  I watched their video so many times it just annoyed me more than it should.  Don't understand how you are supposed to go through the z-axis offset, or the other offsets needed for the configuration, before you install the hardware.  Don't know who edited the video, but the steps are out of order, making it very difficult to follow.  I was expecting the 3DTOUCH to replace my proximity sensor, and it would take over the handling of homing the z-axis.  I must have asked or explained it 3 or 4 times, and got no reply on if this probe even does this.  This morning I found a setting in the configuration.h file that is not enabled by default, "#define USE_PROBE_FOR_Z_HOMING"  Enable that one line and it all of a sudden works as I was expecting.  I know it is something I should have probably found on the first day, and is something that should actually be in the video, or, on the tip of supports tongue if they don't want to document it.  Personally, I cannot understand how any company that is selling items that have a configuration section for it to work properly don't have a simple one page text file with instructions.  It cannot take any longer than 10 minutes to create the document, and would reduce their support budget.  Developers should be creating documentation as part of their job.  

One thing I was told was that the MKS SGEN will not work with TMC2209 drivers, not even the MKS TMC2209 drivers.  I hope this was something the support guy heard about due to this thread, and wasn't something he had known for years.

doug-scott, 2020-10-20:
> Works for me now. Just don't use the UART for tmc220x and that will solve your problem. I'm going to switch to the Sgen L.

They sent me the information on how to get UART to work with the SGEN_L V2.0 with the MKS TMC2209 v2.0 drivers.  Did as they instructed and the TMC COMMUNICATION ERROR went away.  The instructions in the WIKI are mostly correct for the UART enabling, I just found the lack of a simple little diagram showing where to place jumpers made it difficult to figure out where to actually place the jumpers.  The pin layout under the drivers is a simple 3 pin by 4 pin layout.  All they need to do is draw 12 circles in that same 3 x 4 pattern, and then draw a red line connecting the 2 pins needed to enable the setting desired.  It looks like those 12 pins are used to set UART, microstep numbers and enable sensorless homing.  
Note that after setting UART for the drivers it messed up my steps per unit numbers dramatically.  Had to double all the numbers after enabling UART.

URL for WIKI is https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/Drivers_MKS_TMC2209

rpweaverjr, 2025-02-04:
> #define USE_PROBE_FOR_Z_HOMING

I can not find this line in Configuration.h in Marlin-bugfix-2.0.x for MKS SGEN config. Do you have a line number or a different version of Marlin? I don't see a reference to it either.
I am currently using P4_28 and z_max (P1_29) for the 3DTouch
I don't really want to use USE_PROBE_FOR_Z_HOMING but the switch (z_min, P1_28) is disabled during Leveling (G29) for some unknown reason. The switch works for Homing (G28). After the print head attempts to drill through the build surface, I get:

SENDING:G29
Error:Probing Failed
←[01m←[32mcom7←[37m 19°>←[00m Error:Probing Failed

The probe deploys and turns all blue. During the probing, the red led lights but the printhead continues to move after the probe is triggered (is it? the red LED lights).
Everything(?) else seems to be working M401 deploys the probe and extinguishes the red LED, M402 stows the probe and turns the red LED on.
Here is what I see in Pronsole:

com7 19°> M402
SENDING:M402
com7 18°> m119
SENDING:M119
Reporting endstop status
x_min: open    [37m 19°>←[00m
y_min: open    [37m 19°>←[00m
z_min: open    [37m 19°>←[00m
z_max: TRIGGERED37m 19°>←[00m
z_probe: TRIGGEREDm 19°>←[00m
←[01m←[32mcom7←[37m 19°>←[00m

and

com7 19°> m401
SENDING:M401
com7 19°> m119
SENDING:M119
Reporting endstop status
x_min: open    [37m 19°>←[00m
y_min: open    [37m 19°>←[00m
z_min: open    [37m 19°>←[00m
z_max: open    [37m 19°>←[00m
z_probe: open  [37m 19°>←[00m
←[01m←[32mcom7←[37m 19°>←[00m