makerbase-mks/MKS-SGen #16: Configuring BLTouch/3DTouch on the MKS SGEN V 1.0 (not SGEN-L)

rpweaverjr, 2024-12-16:
There don't appear to be servo pins on this board and it seems all the configurations for the BLTouch use the servo pins. What other pins can work on this board in place of the servo pins and what is the associated Pins*.h files

Tinoky, 2025-02-10:
Hello ! I'm actually working on this topic for my DIY printer. 
The pity on this board is that they are using 3 of the PWM to just drive status LED's.

So what I did is I have disabled the **status LED 1**, and then I have connected it's signal **to the NC pin of J7**.
Doing so I can plug my BLTouch on top of **J7**. But pay attention that in order to do that I had to **swap +5V and signal pins on the BL-Touch connector**.

This solution should avoid of using J8 pins that are supposed to be used for TMC drivers UART mode.

On the firmware side, I'm not fully satisfied by the solution I found because I could not make it work from Configuration.h neither from Configuration_adv.h. I had to directly tinker the pin description file **Marlin/src/pins/lpc1769/pins_MKS_SGEN.h** by adding few lines as follow:

```
// Disable Marlin from using this LED for the status.
#undef LED_PIN

//Map SERVO0 to LED1 pin
#undef SERVO0_PIN
#define SERVO0_PIN P1_18
```

It seams to work perfectly on the table, but I didn't check yet on real time on my machine because it is still under construction.
If one of you got a better and cleaner solution, please let me know.

Best regards.

rpweaverjr, 2025-02-12:
I believe I have a concept of what is going wrong with my solution. Unfortunately,  I made some changes and I haven't been able to get back to where I was. That is what I am currently trying to do.
My solution didn't involve disabling anything.  I just wired the prove servo to P4_28 (servo 3) and the probe stop on to z_max. The problem i found may have something to do with "z fade" but I haven't figured out how this works yet.  It seems that if you step a -z small enough,  the movement will stop by the probe being triggered and subsequent -z moves will be ignored. The G29 seems to make longer steps and tries to drill through the print bed. This is because the probe is not interrupt driven and it will do the whole move command before checking the probe status in an infinite loop. If the -z move is larger than -0.1mm it will eventually crash into the build plate unless you are lucky and it happens to stop at 0.1mm after the probe is triggered.
I haven't been able to build any newer Marlin versions to check that this has not already been corrected. If you have a working build of 2.1.x for the SGEN, I'd love to see it.
I bought a new 500W heated bed that has a 110V relay and neodymium magnets and a 60W hot end heater.  I swapped these two drivers because I didn't want to burn out the hot end driver. I don't know what the current limit is for the hot end. That is the only thing I did that is a little different.

Thanks,
Richard Weaver

Sent from my Verizon, Samsung Galaxy smartphone

rpweaverjr, 2025-02-12:
Unfortunately,  my 3D TOUCH seems to be broken. All this messing around and it no longer respond to command and it doesn't go through the startup routine when powering up.  It shows as open regardless of the pin position. It is possible I broke a wire but I don't think so. Unplugging it shows as triggered. I did find a setting that sets the overshoot for the z min probe. It was set to -2mm. It won't accept fractions so I set it to -1mm but I can't test it because the probe is no longer working.  I think I'll try another cable.
Regards,
Richard Weaver

Sent from my Verizon, Samsung Galaxy smartphone

rpweaverjr, 2025-02-15:
I don't know if you are reading these.  I managed to get 2.1.x to build. However, the stepper drivers aren't working. You seem to know something about them. The USB serial port works and the endstops are working. The TMC2209s work in 2.0.x bugfix. I've compared the Configuration.h and ConfigurationAdv.h files between them but much of it doesn't make sense to me. If you would like to help get 2.1.x working fully, let me know.  I'll send you the files.
Thanks,
Richard Weaver

Sent from my Verizon, Samsung Galaxy smartphone